fsstudio-team/ZeroSimROSUnity #3: Ros_bridge not publishing ROS messages

EdvardGrodem, 2021-06-24:
I've followed the tutorial on Setting up a new Unity Project, but want to set up the a new catkin_ws independent of the provided docker space. I'm quite new to ROS so this might be obvious. 
I've tried to run
`roslaunch rosbridge_server rosbridge_tcp.launch bson_only_mode:=true`
And then press play in Unity.
The console in Unity then reports that connection to ros bridge is established, and `rostopic list` confirms that the topics have been published. However, when opening up rviz to view the sensors, none of them receive any messages. 

Is it possible to get access files in the docker container so that it is easier to set up a workspace outside of docker?
Does anyone know how to solve this issue?

EdvardGrodem, 2021-06-28:
In a pivate mail with @micahpearlman  Micah suggested the following:

> Reproducing catkin_ws can be some of the trickiest things about ROS and is why we have released a Docker.  We have not gotten around to open sourcing the docker build files, but will do so soon.  In the meantime see attached docker files.  Probably the part that will fix your `roslaunch rosbridge_server` issue is that there is an issue with the latest ROS Bridge Suite.   The creation of the ROS bridge module for ROS is done via:
> 
> ```
> # install ROS bridge from source
> # NOTE:  there seems to be issues with ROS bridge > 1.11.10 so we use the specific branch
> RUN git clone -b 0.11.10 --single-branch https://github.com/RobotWebTools/rosbridge_suite.git
> RUN cp -R ./rosbridge_suite/rosbridge_suite $CATKIN_WS/src
> RUN cp -R ./rosbridge_suite/rosbridge_library $CATKIN_WS/src
> RUN cp -R ./rosbridge_suite/rosbridge_server $CATKIN_WS/src
> RUN cp -R ./rosbridge_suite/rosbridge_msgs $CATKIN_WS/src
> RUN cp -R ./rosbridge_suite/rosapi $CATKIN_WS/src
> RUN rm -rf ./rosbridge_suite
> ```
> 
> Also, the launch files can be browsed in the ZeroSim docker via opneing a bash script and browsing through the `/catkin_ws/src` directory: `docker exec -it my_zerosim_vnc_docker \
> bash -c "source devel/setup.bash ; rosrun turtlebot3_teleop turtlebot3_teleop_key"`.  Attached is the basic Unity editor launch file.

This almost works. Laser scans and poses etc works well. However, images from cameras do not encode correctly and appears blue in Rviz. Does anyone know how to fix this?

micahpearlman, 2021-06-28:
@EdvardGrodem  Regarding images/cameras not showing up I've opened up a new issue: #4

nsrj, 2021-06-30:
Hi @EdvardGrodem, I am having a similar issue as you described above. The difference is that I could not see any topics published from Unity after running "rostopic list".

I have rosbridge_server 0.11.13 installed with ROS Noetic in my Ubuntu 20.04 machine. After reading the issue here, I downloaded 0.11.10 and catkin_make it in a local workspace. After running "roslaunch rosbridge_server rosbridge_tcp.launch bson_only_mode:=true" and "rostopic list" on two consoles, I still could not see any topics from Unity. 

Could you let me know what you did to make this work, please? Thanks a lot.

EdvardGrodem, 2021-07-01:
Hi, @nsrj.  The [Recomended System](https://github.com/fsstudio-team/ZeroSimROSUnity#recommended-system) specifies ROS Melodic, however I'm not familiar enough with ROS to be able to tell if Noetic could work.  
I am running Ubuntu 18.04 with ROS Melodic and ROS bridge 0.11.10. Other then that I think we have the same setup.

micahpearlman, 2021-07-01:
@nsrj you may want to checkout this pull request for ROS Bridge that hasn't been incorporated: https://github.com/RobotWebTools/rosbridge_suite/pull/545

nsrj, 2021-07-01:
@micahpearlman, thank you for your follow-up. I downloaded this [version](https://github.com/nick-paul/rosbridge_suite/tree/bson-py3) and it is now working on my machine. The issue is the rosbridge. Thanks a lot for pointing it out!

nsrj, 2021-07-01:
@EdvardGrodem, thank you for your reply. After switching to a different version of rosbridge as mentioned above, all the topics are visible now. Thanks!

cramirez98, 2021-09-19:
> Hi @EdvardGrodem, I am having a similar issue as you described above. The difference is that I could not see any topics published from Unity after running "rostopic list".
>
> I have rosbridge_server [...]

Hello, im having the same problem, i tried downloading and compiling this [ROSBridge version](https://github.com/nick-paul/rosbridge_suite/tree/bson-py3), but when i run:
> roslaunch rosbridge_server rosbridge_tcp.launch bson_only_mode:=true

i cant see any topics from Unity, i really dont know that else to try, i can perfectly see the topics from Unity being published when i use the docker catkin workspace, but when i try to see them from my local catkin workscape using **_rostopic list_**, i cant see them and when i check the Unity console, it reports that connection to ros bridge is established. Im running ROS Noetic on a Ubuntu 20.04 machine.

micahpearlman, 2021-09-19:
@cramirez98 this is the ZeroSim Docker setup for ROS Bridge:

```
# install ROS bridge from source
# NOTE:  there seems to be issues with ROS bridge > 1.11.10 so we use the specific branch
RUN git clone -b 0.11.10 --single-branch https://github.com/RobotWebTools/rosbridge_suite.git
RUN cp -R ./rosbridge_suite/rosbridge_suite $CATKIN_WS/src
RUN cp -R ./rosbridge_suite/rosbridge_library $CATKIN_WS/src
RUN cp -R ./rosbridge_suite/rosbridge_server $CATKIN_WS/src
RUN cp -R ./rosbridge_suite/rosbridge_msgs $CATKIN_WS/src
RUN cp -R ./rosbridge_suite/rosapi $CATKIN_WS/src
RUN rm -rf ./rosbridge_suite

```
See: https://github.com/fsstudio-team/zerosim_docker/blob/master/docker/base/Dockerfile

micahpearlman, 2021-09-21:
@cramirez98 any luck?  If so, do you mind sharing how you were able to setup a non-Docker system for ROS Noetic?

cramirez98, 2021-09-21:
No, i couldn't get it to work

micahpearlman, 2021-09-23:
@cramirez98 perhaps you can try running ROS Bridge in the ZeroSim Docker and run other ROS nodes on your host OS by juggling the ROS master environment settings: https://answers.ros.org/question/272065/specification-of-ros_master_uri-and-ros_hostname/

leander-dsouza, 2021-09-27:
I have managed to set up the ROS Bridge mentioned **[here](https://github.com/nick-paul/rosbridge_suite/tree/bson-py3)** on my system (ROS Noetic, Ubuntu 20.04 LTS), and the topics from Unity were visible.

When I tried to set up the default **[rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite/tree/ros1)** package from apt, the server used to form a connection, but there were no topics from Unity being transferred. So the fix mentioned by @micahpearlman indeed does work on Noetic as well.

@cramirez98 I know this seems trivial, but were you able to confirm that the ROS path for the `rosbridge_suite` package was from your workspace and not from `/opt/ros/noetic`? An excellent way to check the path is using the `roscd rosbridge_suite` command.

alex-ssom, 2022-04-15:
I found the easy way to solve this issue, but it only works on a linux host.

Add the `--network host` argument to the `docker run` command. You can also remove the `--publish` arguments to prevent the getting `WARNING: Published ports are discarded when using host network mode`.

# Proof of fix
## Run the docker image
```bash
docker run -it --rm \
--network host \
--name my_zerosim_vnc_docker \
zerodog/zerosim_ros_vnc:latest \
roslaunch zero_sim_ros basic_unity_editor.launch
```

## rosbridge 0.11.13 (ROS Noetic, Ubuntu 20.04 LTS)
```bash
➜  catkin_ws roswtf
Loaded plugin tf.tfwtf
No package or stack in the current directory
================================================================================
Static checks summary:

No errors or warnings
================================================================================
Beginning tests of your ROS graph. These may take a while...
analyzing graph...
... done analyzing graph
running graph rules...
... done running graph rules
running tf checks, this will take a second...
... tf checks complete

Online checks summary:

Found 1 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING No tf messages

➜  catkin_ws rosversion rosbridge_server
0.11.13
➜  catkin_ws rostopic echo /client_count 
data: 0
---
```