fsstudio-team/ZeroSimROSUnity #4: Image Sensor Not Rendering

micahpearlman, 2021-06-28:
From: @EdvardGrodem in issue #3  

images from cameras do not encode correctly and appears blue in Rviz. Does anyone know how to fix this?

micahpearlman, 2021-06-28:
I suspect that this is an issue with the underlying Unity rendering driver.  The image/camera renders require Vulkan drivers (OpenGL will likely not work).  Can you verify that the Vulkan driver is setup properly as outlined in: https://github.com/fsstudio-team/ZeroSimROSUnity#setting-up-a-new-unity-project

micahpearlman, 2021-06-28:
I just tested OpenGL on my system.  It does render and I can visualize in RViz but the colors are incorrect.  Are you sure the camera is attached to something in the scene and not falling through the floor or pointed in a unexpected direction?  Can you try to set the debug parameter in the ZORGBCamera compontent, see image below.  The camera sensor image should be rendered when Unity is running (it will be upside down, that is okay).

[image]

EdvardGrodem, 2021-06-29:
Yes, enabling Vulken fixed the issue! I had some stability problems and forgot that I turned it off to test if that improved stability. Thanks a lot!

closed 2021-06-29 by micahpearlman