fsstudio-team/ZeroSimROSUnity #5: ROS publisher for 3D Lidar

nsrj, 2021-06-28:
Hi, thanks for sharing the project. The TurtleBot Test sample works very well. A 2D lidar is used in the sample. I'd like to know the way to use the 3D lidar. I couldn't find the ROS publisher for the 3D lidar. Could you advise on this, please?
Thank you!

micahpearlman, 2021-06-29:
@nsrj we have not yet implemented a publisher for 3D LIDAR yet. When we do what would you consider to be the preferred ROS message for 3D LIDAR? PointCloud or PointCloud2?

nsrj, 2021-06-29:
@micahpearlman, thanks a lot for the reply. PointCloud2 is preferred. But PointCloud is also fine. I just need to do a conversion on my own. 

Would you have a timeline for implementing the publisher for 3D lidar? I am looking forward to it. 

Thank you again.

micahpearlman, 2021-06-29:
@nsrj I just implemented a first version of the `PointCloud2` Publisher that will publish the XYZ components of the 3D LIDAR.  You can see an example in a new test scene: `Scenes/ROSPointCloud2Publish_test.unity` 

NOTE:  Updating the ZeroSim package is not automagic in Unity for whatever reason.  You will want to do the: Go into the package manager, select "+" button, and then select `Add Package From Git URL...` and enter `git@github.com:fsstudio-team/ZeroSimROSUnity.git` again.  And the re-import the ZeroSim samples to get the above mentioned test scene.  The package version should be `0.1.9`

nsrj, 2021-06-30:
@micahpearlman thanks a lot for your quick reply and updates on the project. I have run the ROSPointCloud2Publish_test.unity, the point cloud looks very good. Thank you for the great work!

I have a follow-up question. Would it be possible to launch rosbridge_tcp.launch natively on the command line instead of launching it in Docker? Could you advise on this, please? Thank you again.

micahpearlman, 2021-06-30:
@nsrj take a look at issue #3 regarding ROS Bridge.  Attached is the ZeroSIm ROS docker build file.
[Dockerfile.txt](https://github.com/fsstudio-team/ZeroSimROSUnity/files/6743231/Dockerfile.txt)

>     Reproducing catkin_ws can be some of the trickiest things about ROS and is why we have released a Docker. We have not gotten around to open sourcing the docker build files, but will do so soon. In the meantime see attached docker files. Probably the part that will fix your roslaunch rosbridge_server issue is that there is an issue with the latest ROS Bridge Suite. The creation of the ROS bridge module for ROS is done via:
> 
>     # install ROS bridge from source
>     # NOTE:  there seems to be issues with ROS bridge > 1.11.10 so we use the specific branch
>     RUN git clone -b 0.11.10 --single-branch https://github.com/RobotWebTools/rosbridge_suite.git
>     RUN cp -R ./rosbridge_suite/rosbridge_suite $CATKIN_WS/src
>     RUN cp -R ./rosbridge_suite/rosbridge_library $CATKIN_WS/src
>     RUN cp -R ./rosbridge_suite/rosbridge_server $CATKIN_WS/src
>     RUN cp -R ./rosbridge_suite/rosbridge_msgs $CATKIN_WS/src
>     RUN cp -R ./rosbridge_suite/rosapi $CATKIN_WS/src
>     RUN rm -rf ./rosbridge_suite
> 
>     Also, the launch files can be browsed in the ZeroSim docker via opneing a bash script and browsing through the /catkin_ws/src directory: docker exec -it my_zerosim_vnc_docker \ bash -c "source devel/setup.bash ; rosrun turtlebot3_teleop turtlebot3_teleop_key". Attached is the basic Unity editor launch file.
>

nsrj, 2021-06-30:
@micahpearlman, thank you for your reply.
I have 0.11.13 installed on my machine. By following your instruction, I downloaded 0.11.10 in the local workspace and catkin_make the package. After running "roslaunch rosbridge_server rosbridge_tcp.launch bson_only_mode:=true" on a terminal, and running "rostopic list" on another terminal, I could not see any topics from Unity. I have Ubuntu 20.04 and ROS-Noetic installed on the machine. I wonder if this is the problem.

Thanks a lot for your help!

micahpearlman, 2021-06-30:
ROS installation issue are notoriously difficult to debug. My only guess is somehow the wrong version of ROS Bridge is being used or there is a Noetic specific issue?  For ROS I always highly recommend using Dockers otherwise you can chase your tail. You may want to reach out to the person who reported #3 and see how they solved this issue? If you do find a solution please report back here!

closed 2021-07-05 by micahpearlman