fsstudio-team/ZeroSimROSUnity #6: RGBD camera mirrored

EdvardGrodem, 2021-07-05:
Hi, I've been checking out the depth camera module. Sometimes it works fine, however other times the image is mirrored. This is how I recreated the issue:
Open ZeroSim samples/ROSDepthImagePublish_test
Run ROS and play scene
Open Rviz
And everything works fine.

Then I make a prefab of the RGBDepthCameraSensor GameObject. 
Open a new scene, add the RGBDepthCameraSensor prefab. Add a ZOROS Unity Manager with WorldFrameId = map
Open ROS, play scene
Open Rviz
The depth image is mirrored.

Is this bug, is there something I don't get?
Thanks
Edvard

micahpearlman, 2021-07-05:
@EdvardGrodem there does seem to be some weirdness that we are looking into.  No ETA quite yet.

micahpearlman, 2021-07-05:
Okay, the behavior of the RGBD sensor and publisher should be more consistent.  You will need to grab the latest version.

NOTE: Updating the ZeroSim package is not automagic in Unity for whatever reason. You will want to: Go into the Unity Package Manager, select "+" button, and then select `Add Package From Git URL...` and enter `git@github.com:fsstudio-team/ZeroSimROSUnity.git` . Finally re-import the ZeroSim samples. The ZeroSim Unity package version should be 0.1.10

EdvardGrodem, 2021-07-06:
That fixed the issue! Thank you so much for the fast response time! 

A small note: I had to use `https://github.com/fsstudio-team/ZeroSimROSUnity.git` in the Package Manager to be able to import the package

closed 2021-07-06 by micahpearlman