fsstudio-team/ZeroSimROSUnity #7: Unity Left Handed Coordinate System

EdvardGrodem, 2021-07-09:
Hi, after trying to run SLAM from the simulator we have some trouble with the coordinate frames from Unity. I know Unity uses a left handed coordinate system. Does ZeroSim account for this when publishing transforms?

EdvardGrodem, 2021-07-09:
From what I can tell form the code the conversion is implemented, but not used by the transform publisher. 
https://github.com/fsstudio-team/ZeroSimROSUnity/blob/ead163a96296c3b3dbd83315e050941c259b3892/Runtime/Scripts/ROS/MessageTypes/Geometry/ZOROSGeometryMessages.cs#L36-L67

micahpearlman, 2021-07-09:
So I tried this out by creating a simple scene with a TF Publisher with coordinates (1, 2, 3) and looked at the result in RViz and the TF position is (3, -1, 2) as expected.  See attached images.

We have run ROS gmapper successfully.  What SLAM system are you using?

[image]
[image]

micahpearlman, 2021-07-28:
@EdvardGrodem are you still having issues?  Found a major issue with converting Unity Quaternion to ROS.  v0.1.14 should fix that.

EdvardGrodem, 2021-08-03:
Hi, we are still struggeling with running SLAM, however I do not think it's due to the left handed coordinate system anymore. I think the initial confusion was due to the cameras not having the z-axis pointing into the image as is common convention. We found the settings for correcting this.

micahpearlman, 2021-08-03:
@EdvardGrodem can you elaborate on the camera coordinate system and how it was fixed?  Is there a patch or pull request that you can provide?

EdvardGrodem, 2021-08-30:
Hi, sorry for not coming back to this sooner. In ZOROSRBGDepthPublisher we used the Camera Rotation Degrees to align the camera with our preferred coordinate system.

szandara, 2021-12-16:
On this note, I am struggling to understand the vector conversion. I am expecting 

FromUnityVector(ToUnityVector3([0.1, 0.2, 0.3])) == [0.1, 0.2, 0.3] but this is not true. 

`ss`