fsstudio-team/ZeroSimROSUnity #8: Point clouds of the 3D lidar have no difference when the orientation of the transform is changed

bj108, 2021-07-28:
Hi @micahpearlman, I am using the 3D lidar for object detection. I noticed that the point clouds of the lidar have no difference when the orientation (rotation) of the transform is changed but have differences when the position is changed.

I am with ZeroSim 0.1.9 and running the ROSPointCloud2Publish_test scene.

Are there any parameters for the lidar or publisher needed to be set when changing the orientation? Could you please give some advice? Thank you!

micahpearlman, 2021-07-28:
@bj108 Thank you for reporting this bug.  We have fixed this issue.  If you can try out version 0.1.15.

NOTE: For whatever reason, updating the ZeroSim package is not automagic in Unity. You will want to: Go into the package manager, select "+" button, and then select `Add Package From Git URL...` and enter `https://github.com/fsstudio-team/ZeroSimROSUnity.git` again.  The package version should be 0.1.15

bj108, 2021-07-29:
Hi @micahpearlmanm, thank you for the quick fixing. I am trying out version 0.1.15. The point clouds look different now when the orientation is changed.

I'd like to bring up another question, please look at the attached screenshots below:

[image]
[image]
[image]
[image]

The rotation around the X-axis of the Transform was changed from 0 to 15, 30, and 45 degree. In the rviz, since the lidar frame rotated clockwise, the point cloud of the cylinder and cube should rotate counterclockwise. But in fact, the point cloud of the cylinder and cube did not rotate accordingly. Could you advise on this, please? Thank you.

micahpearlman, 2021-07-29:
@bj108 I think that may be an illusion because of the vertical spread of the LIDAR is so large.  If you reduce the vertical FOV spread you can see more clearly the rotation.
[image]
[image]
[image]
[image]
[image]

bj108, 2021-07-30:
Hi @micahpearlman, thank you for your quick answer.

I'd like to follow up on a question. Please see the below screenshot.

[image]

As can be seen, when the lidar rotates 20 degree around the X-axis, the point cloud also rotates 20 degree clockwise. This means the lidar frame rotates 20 degree with respect to the RGB frame shown in the Rviz. My question is, can the lidar frame not rotate (be the same as the RGB frame all the time) and the point cloud rotates 20 degree counterclockwise?

In addition, could you please explain the Frame Id and Child Frame Id in the ZOROS Transform Publisher and the Parent Transform Id in the ZOROS Point Cloud 2 Publisher? Thanks a lot!

micahpearlman, 2021-07-30:
@bj108 I'm not sure I'm following what the issue is?  I'm not clear on what you mean by RGB frame?  You can make the LIDAR frame relative to any frame you want, in the `ROSPointCloud2Publish_test` it is relative to the `map` frame (aka fixed world frame defined in  `ZOROSUnityManager`).

the `ZOROSTransformPublisher` implements the ROS TF transform system (http://wiki.ros.org/tf).  It publishes the local Unity `Transform` for the Transform that the script is attached to.  The `ChildFrameID` is the ID/name of the current transform and `FrameID` is the parent transform name -- terrible naming I know but this is ROS convention (http://docs.ros.org/en/melodic/api/geometry_msgs/html/msg/TransformStamped.html).

bj108, 2021-08-03:
@micahpearlman, thanks a lot for your response. I am writing to provide more information about this issue.
In the TurtleBot3_Waffle_test, the 2D lidar rotates 5 degree around the X-axis in Unity (CCW in the screenshot), while in the rviz, the point cloud rotates CW, which seems correct.
[image]
[image]

In the ROSPointCloud2Publish_test, the 3D lidar rotates -10 degree around the X-axis in Unity (CW in the screenshot), while in the rviz, the point cloud also rotates CW, which does not seem right. But I have no idea how to fix this issue.

[image]
[image]

micahpearlman, 2021-08-03:
@bj108 I'm not sure if I'm following exactly what the issue is so easily in the images and description you have provided.  The big difference I see in the images that you have provided is that in the Turtlebot scene RViz Fixed Frame is set to `scan` (which is the LIDAR frame) while in the `ROSPointCloud2Publish_test` the RViz Fixed Frame is set to `map` which is the world frame.

bj108, 2021-08-03:
@micahpearlman, thank you for your quick response. Please allow me to ask this question in a different way. Let's say, the point cloud of the 3D lidar is published with a frame_id "lidar_frame". In the Rviz, the Global Options --> Fixed Frame is also selected to "lidar_frame". Furthermore, the Axes (see below figure) are added to the Rviz Grid.
[image]
If rotating the lidar in Unity along an axis (for example, positive X), since the lidar_frame is the same as the Axes in Rviz, the lidar point cloud should rotate in the opposite direction (negative X) in Rviz. 

I tried changing the Frame Id and Child Frame Id in the ZOROS Transform Publisher and the Parent Transform Id in ZOROS Point Cloud 2 Publisher but could not make it work.

micahpearlman, 2021-08-03:
@bj108 apologies but I’m still not following. From my experiments rotation about the X in Unity appears correct in RViz and unfortunately I don’t   clearly from your description — I’m sure there is something obvious I’m missing?  Can you also tell me what version you are using?  The latest version does have a major fix in rotation and perhaps that is the issue?

bj108, 2021-08-03:
@micahpearlman, I appreciate your help! I am using 0.1.15. Could you please explain how to set up the **Frame Id** and **Child Frame Id** in the **ZOROS Transform Publisher** and the **Parent Transform Id** in **ZOROS Point Cloud 2 Publisher**?

micahpearlman, 2021-08-04:
`ZOROSTransformPublisher` is a part of ZeroSim to interface with the ROS TF Broadcast system.  The TF Broadcast system maintains relationship between parent and child coordinate frames overtime.  For example a robot navigating the world with a LIDAR attached:

```
[map] ---> [base_link] ---> [lidar_link] 
```
`map` is the default fixed world coordinate frame and defined in `ZOROSUnityManager World Frame Id`. (`map` for the fixed world frame is a ROS convention but it can be named whatever you want).

`base_link` is the robot main body ('base_link' is a ROS convention but it can be named whatever you want).  If you were to create a robot in a Unity scene you would add a `ZOROSTransformPublisher` to the root body of the robot and assign `Frame Id` to be `map` and `Child Frame Id` to be `base_link`.

`lidar_link` is relative to the `base_link` so it's `ZOROSTransformPublisher` would have a `Frame Id` of `base_link` and a `Child Frame Id` of `lidar_link`.  It would also have a `ZOROSPointCloud2Publisher` and it's `Parent Transform Id` for almost all cases should be the parent id, in this case `base_link`.  The  `ZOROSPointCloud2Publisher` uses the `Parent Transform id` in the `header` portion of the ROS `PointCloud2` message (http://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/PointCloud2.html).  A user can change the `Parent Transform Id` to whatever they want be I cannot think of an instance of where it shouldn't be the parent ID or the world fixed frame ID.  When  `ZOROSPointCloud2Publisher` is created in a scene it trys it's best to figure out what the best default value should be by searching for a parent `ZOROSTransformPublisher` and if not found getting the `ZOROSUnityManager World Frame Id`.

micahpearlman, 2021-08-06:
@bj108 did the above explanation help with the understanding of the ZeroSim TF?  I am still not able to quite understand what your current issue is?

bj108, 2021-08-07:
Thank you. Your detailed explanation helps a lot! But I still don't have the luck making the 3D point cloud work. I'd like to give an example here. I am holding a lidar horizontally and a body-fixed frame is attached to the lidar. A wall is in front of the lidar. In Rviz, selecting the fixed-frame as the lidar body-fixed frame, you will see a point cloud of the wall in front of the lidar frame. If the lidar rotates 90 degree down to the ground, you will see a point cloud of the ground in front of the lidar frame. Note that since the fixed-frame is selected to the lidar body-fixed frame, so this frame won't move in Rviz but the point cloud rotates 90 degree up. 
I guess there may be an issue with the transform between the map and the lidar frame.

micahpearlman, 2021-08-07:
@bj108 apologies, but again either I don’t understand the issue or I cannot reproduce anything that looks incorrect?  Is there anyway you can set a test case scene and rviz setup?

bj108, 2021-08-10:
Thank you. What you described above is totally correct. Could you also explain the role of the lidar_link? Does it matter in the navigation when the base_link is the Parent Transform Id in the Point Cloud 2 Publisher?

One more question is the Vertical Up and Down Fov Degrees. If both of them are set to 10 degree. Does it mean that the vertical FOV is 20 degree, 10 degree from horizontal up and 10 degree from horizontal down? Thank you!

micahpearlman, 2021-08-10:
I'm not sure I understand your question regarding the `Parent Transform Id` and `base_link` in regards to navigation?  Generally the `base_link` is something like the robot chassis that the LIDAR sensor is physically attached to.

Yes, the total vertical FOV is the sum of the high and low FOVs.

closed 2022-04-04 by micahpearlman