fsstudio-team/ZeroSimROSUnity #9: IMU publisher inconsistent coordiante system

EdvardGrodem, 2021-08-03:
The description of this IMU configuration yields a left handed coordinate system but claims to be right handed. 

https://github.com/fsstudio-team/ZeroSimROSUnity/blob/ead163a96296c3b3dbd83315e050941c259b3892/Runtime/Scripts/ROS/Unity/Publishers/ZOROSIMUPublisher.cs#L131

micahpearlman, 2021-08-03:
@EdvardGrodem thank you for the bug report.  Can you provide more detail on how you are testing & verifying this issue?

rendellc, 2021-08-04:
Coworker of @EdvardGrodem here, we didn't use this setting so we didn't run into this bug ourselves. We only found it by chance because we were reading the source code to get a better understanding of what ZeroSim does

micahpearlman, 2021-08-04:
@rendellc thank you for the clarification. Any chance we can get a patch or pull request?