fsstudio-team/ZeroSimROSUnity #11: Import does not fetch git lfs files

rendellc, 2021-08-11:
When creating a new project, the import does not fetch lfs files. This causes several error messages, which I have pasted below for searchability. Opening the files, for instance `Runtime/Scripts/Sensors/RGBDepthCamera/calibration_chessboard.png`, in a text editor it is clearly a git-lfs file.
```
version https://git-lfs.github.com/spec/v1
oid sha256:36a04f9701b424550dbff7cf8c4e5cd7a12e68cea45a98774285af7799492254
size 62550
```
Looking at [Unity git guide](https://docs.unity3d.com/Manual/upm-git.html),  I would expect this to be handled by Unity so I am not sure what causes this issue. It may be a system configuration issue on my end. 

System:
- Ubuntu 18.04
- git version 2.17.1
- git-lfs version 2.3.4

**Temporary workaround**
1. Add package manually by cloning repo into unity project.
2. Run `git lfs fetch --all` and `git lfs checkout .` to retrieve the lfs files.

**Some of the errors**
```
Library/PackageCache/com.fsstudio.zerosim@cfa659082a/Runtime/Scripts/Util/Docker/ZODocker.cs(9,7): error CS0246: The type or namespace name 'Docker' could not be found (are you missing a using directive or an assembly reference?)
```
```
Could not create asset from Packages/com.fsstudio.zerosim/Runtime/Scripts/Sensors/RGBDepthCamera/calibration_chessboard.png: File could not be read
```
```
ImportFBX Errors:
Couldn't read file /home/rendell/UnityProjects/ZeroSim test/Assets/Samples/ZeroSim/0.1.15/ZeroSim samples/Assets/Models/Table.fbx.
```

rendellc, 2021-08-12:
Using `https://github.com/fsstudio-team/ZeroSimROSUnity.git` gives the same results. It is likely a configuration issue on my end. I will retry on another computer with another Unity setup

rendellc, 2021-08-12:
Closing issue as I got it to work now. I had to set some lfs settings manually. Don't think all these settings are relevant, but for my current settings are (obtained by `git config --list`).
```
lfs.dialtimeout=90
lfs.tlstimeout=90
lfs.batch=true
lfs.contenttype=0
filter.lfs.process=git-lfs filter-process
filter.lfs.required=true
filter.lfs.clean=git-lfs clean -- %f
filter.lfs.smudge=git-lfs smudge -- %f
```

closed 2021-08-12 by rendellc

florianfhws, 2022-02-22:
For me unity throws  Error when executing git command. git@github.com: Permission denied (publickey).   fatal: Could not read from remote repositor when using git@github.com:fsstudio-team/ZeroSimROSUnity.git however with https://github.com/fsstudio-team/ZeroSimROSUnity.git it works fine