fsstudio-team/ZeroSimROSUnity #15: ZODocker.cs error after installing ZeroSim package for Unity 3D

cramirez98, 2021-09-17:
Hello, i tried to do the "Running TurtleBot Test Scene" tutorial, but when i tried to start the Unity simulation, i see 2 errors on the console, they say the following:

1. Library/PackageCache/com.fsstudio.zerosim@c832f0ecd7/Runtime/Scripts/Util/Docker/ZODocker.cs(9,7): error CS0246: The type or namespace name 'Docker' could not be found (are you missing a using directive or an assembly reference?)
2.  Library/PackageCache/com.fsstudio.zerosim@c832f0ecd7/Runtime/Scripts/Util/Docker/ZODocker.cs(10,7): error CS0246: The type or namespace name 'Docker' could not be found (are you missing a using directive or an assembly reference?)

I really dont know what i did wrong because i did everything that is said in the "Setting up a new Unity Project" and "Getting ZeroSim ROS Docker Container" tutorials. My system is: Ubuntu 20.04, ROS Noetic, Unity 2020.3.18f1.
I already installed docker and can run it without any problem.

micahpearlman, 2021-09-17:
Thank you for spotting this issue.  I'm actually surprised this had not come up much earlier. This is an error that there is a dependency on `Docker.Net` Package (https://github.com/dotnet/Docker.DotNet).  As of the newly release v01.1.17 we have removed that dependency. 

NOTE: Updating the ZeroSim package is not automagic in Unity. You will want to: Go into the Unity Package Manager, select "+" button, and then select Add Package From Git URL... and enter git@github.com:fsstudio-team/ZeroSimROSUnity.git . Finally re-import the ZeroSim samples. The ZeroSim Unity package version should be 0.1.17

cramirez98, 2021-09-17:
Now when i update ZeroSim and re-import the samples i get this error:

> Library/PackageCache/com.fsstudio.zerosim@dff8f64f72/Runtime/Scripts/ROS/ROSBridge/ZOROSBridgeConnection.cs(367,52): error CS0246: The type or namespace name 'BsonDataReader' could not be found (are you missing a using directive or an assembly reference?)

cramirez98, 2021-09-17:
Im closing this issue because I found the problem and it's a problem with the Import mechanism in Unity Package Manager, it seems like it doesn't get the git lfs files, I tried to fix it by cloning the repo manually with git lfs installed and it works. You should put some kind of warning or caution message on the Getting Started tutorial so people who find this problem can know what they can do to fix it. Anyways, thanks for developing this package, its very useful for robotics students like me.

closed 2021-09-17 by cramirez98

micahpearlman, 2021-09-17:
@cramirez98 I'm a bit at a loss.  That should be included as part of the ZeroSim package (see: https://github.com/fsstudio-team/ZeroSimROSUnity/tree/master/Runtime/Plugins)  -- actually just notice `Docker.DotNet` is in there as well .  Can you try installing in a fresh project? If that doesn't work can you possibly try an older Unity version.  The latest we have tested with is 2020.3.14f1.  We will try 2020.3.18f1 but it may be a few working days.

micahpearlman, 2021-09-17:
@cramirez98  Oh excellent.  How did you fix the LFS issue?  That has been reported a couple of times but it's not been clear on how to fix it and we have been unable to reproduce it.

cramirez98, 2021-09-17:
The procedure i did was the following:

1. Install ZeroSim package using the **Add Package From Git URL** option in the PackageManager in Unity (As explained in the Getting Started tutorial).
2. Close your Unity project.
3. Install Git LFS if you haven't installed it before.
4. Go to the directory /YourProjectDirectory/Library/PackageCache/ and there will be a folder with a name like com.fsstudio.zerosim@dff8f64f72 (dff8f64f72 is related with the last commit so it can vary), delete all its content, including hidden files.
5. Clone the repo into the com.fsstudio.zerosim@dff8f64f72  folder with the following git command (You have to cd to the PackageCache folder first):
> $ git clone https://github.com/fsstudio-team/ZeroSimROSUnity.git com.fsstudio.zerosim@dff8f64f72
6. Open your Unity project and import the files using the PackageManager, you will see that the console errors are gone and all the files are loaded properly.

cramirez98, 2021-09-17:
Forgot to say that i cloned the repo using Git LFS v2.13.3

leander-dsouza, 2021-11-30:
I followed the procedure mentioned by @cramirez98, just had to replace `git clone` with ` git-lfs clone` in step 4.

topguns837, 2023-07-25:
@cramirez98 Thanks for the help !