fsstudio-team/ZeroSimROSUnity #21: ROS Camera depth publisher: ROSUnityManager not instantiate

Guillaume-Syc, 2022-03-24:
Hello !
I've just download the package on Unity (ubuntu). I've place the ZORGBDepthCamera on an GameObject with a ZOROSRGBDepthPublisher. I've use the docker project to launch zero_sim_ros basic_unity_editor.launch. 
Unity give me this error: 
`NullReferenceException:` Object reference not set to an instance of an object
ZO.ROS.Publisher.ZOROSRGBDepthPublisher.ZOUpdateHzSynchronized () (at Library/PackageCache/com.fsstudio.zerosim@354853b8f1/Runtime/Scripts/ROS/Unity/Publishers/ZOROSRGBDepthPublisher.cs:113)
ZO.Util.ZOGameObjectBase.Update () (at `Library/PackageCache/com.fsstudio.zerosim@354853b8f1/Runtime/Scripts/Util/GameObject/ZOGameObjectBase.cs:232)`
When I look at the object "ROSUnityManager" in the debug mode, it a null object.
Did I miss a step at the installation or some thing like that ?  
Thank you !

anqitongxue, 2022-03-24:
这里是氨气同学的QQ邮箱，您的邮件我已收到，祝生活愉快~

micahpearlman, 2022-03-24:
No idea.  Did you add a `ZOROSUnityManager` to the scene?

micahpearlman, 2022-03-31:
Any luck?

closed 2022-04-04 by micahpearlman