fsstudio-team/ZeroSimROSUnity #22: Import URDF not working as expected

alex-ssom, 2022-03-31:
HI, 

I'm trying to import the [LeoRover description](https://github.com/LeoRover/leo_common/tree/master/leo_description) in Unity using ZeroSim. I followed the video tutorial and the comment from @micahpearlman in https://github.com/fsstudio-team/ZeroSimROSUnity/issues/17#issuecomment-995344947. I was able to import the `.obj` files but  it looks like the joints weren't imported properly.

[image]

Have I missed something or is this the expected behaviour?

Thanks, 
Alex

micahpearlman, 2022-04-04:
Hmmm, no idea?  Can you share the scene?

micahpearlman, 2022-04-04:
Also, see issue: https://github.com/fsstudio-team/ZeroSimROSUnity/issues/17

alex-ssom, 2022-04-04:
> Hmmm, no idea? Can you share the scene?

I created a new scene then used `ZeroSim -> Import URDF` 
https://github.com/alex-ssom/zerosim_import_leorover_urdf.git

I noticed that the hierarchy tree is different than the one generated when using `ZeroSim -> New Robot`

micahpearlman, 2022-04-05:
@alex-ssom so I was able to import the LEO robot.  There is a full description in the `README.md`.

There was a fix we needed to do so you will want to upgrade ZeroSim to the latest version of 0.1.18.

[image]

#### LEO Robot Example

1. Clone the Leo robot: `git clone --recursive https://github.com/LeoRover/leo_common.git`
2. Checkout Melodic version.  `cd leo_common && git checkout melodic`.
3. Startup ZeroSim Docker mounting the LEO directory:

```

# make sure you are in parent directory for `leo_common`
cd .. 

# run docker
docker run -it --rm \
--publish=9090:9090 \
--publish=11311:11311 \
--publish=8083:8083 \
--publish=80:80 \
--publish=5678:5678 \
--name my_zerosim_vnc_docker \
--volume=$(pwd)/leo_common/:/catkin_ws/src/leo_common/ \
zerodog/zerosim_ros_vnc:latest \
bash
```
4. In the Docker command prompt build the catkin workspace:

```
source devel/setup.bash
catkin build
# Make sure to source ROS again to get the new LEO robot
source devel/setup.bash
```
5. In the Docker command prompt Run XAcro on the LEO robot to get the URDF:

```
rosrun xacro xacro src/leo_common/leo_description/urdf/leo_sim.urdf.xacro > /tmp/leo_sim.urdf
```
6. Convert LEO robot meshes to .obj for Unity.  In the Docker command prompt:

```
/zerosim_tools/convert_meshes_to_obj.sh ./src/leo_common/leo_description/models
```

7. Copy URDF from Docker to host:

```
# create a directory to store the URDF and meshes
mkdir my_leo_robot

# copy the URDF
docker cp my_zerosim_vnc_docker:/tmp/leo_sim.urdf ./my_leo_robot 
```

8. Copy Meshes from Docker to host:

```
# Note we are preserving the path
docker cp my_zerosim_vnc_docker:/catkin_ws/src/leo_common/leo_description/models ./my_leo_robot/leo_description/models
```

9. Fix the URDF paths to the meshes:

```
# First fix up the .DAEs to point to the OBJs
sed -i 's#.dae#.obj#g' my_leo_robot/leo_sim.urdf

# The fix up the .STLs to point to the OBJs
sed -i 's#.stl#.obj#g' my_leo_robot/leo_sim.urdf

# Now remove the `package://` because we are using the filesystem
sed -i 's#package://#./#g' my_leo_robot/leo_sim.urdf 

```

10. In Unity import URDF by: Right click and select `ZeroSim --> Import URDF...`

micahpearlman, 2022-04-06:
@alex-ssom do us a favor, if necessary, fix or add any notes for the instructions above -- this will help a lot of users and be a big help for us.

alex-ssom, 2022-04-06:
### Testing the provided instructions
The provided instructions worked like a charm except for step 7 and 8. 
Step 7:
```bash
# create a directory and a sub-directory to store the URDF and meshes
mkdir -p my_leo_robot/leo_description
```

Step 8: The folder should be removed from the second path.
```bash
# Note we are preserving the path
docker cp my_zerosim_vnc_docker:/catkin_ws/src/leo_common/leo_description/models ./my_leo_robot/leo_description
```

### Using an extended URDF
It worked so well in fact that I tried using my xacro add on package.

1) I needed to add a second volume:
```bash
docker run -it --rm \
--publish=9090:9090 \
--publish=11311:11311 \
--publish=8083:8083 \
--publish=80:80 \
--publish=5678:5678 \
--name my_zerosim_vnc_docker \
--volume=$(pwd)/leo_common/:/catkin_ws/src/leo_common/ \
--volume=$(pwd)/leo_ssom_addons_description/:/catkin_ws/src/leo_ssom_addons_description/ \
zerodog/zerosim_ros_vnc:latest \
bash
```
2) Install the realsense2_description package
```bash
sudo apt update && sudo apt install ros-melodic-realsense2-description
```
3) Step 4 and 5 as is.

4) Run step 6 for all packages used by my URDF.
5) Create a new directory in `my_leo_robot` for all additional packages.
6) Run step 8 for all packages used by my URDF.
7) Step 9 as is.

The only issue I encountered is that the add on objects aren't localised and oriented properly. Probably because they haven't been drawn using the same CAD software therefore their original orientations aren't identical. This doesn't happen when displaying my URDF using RViz.

[image]

I manually placed some of said objects.
[image]

micahpearlman, 2022-04-06:
Fantastic!  We appreciate your patience in working with us to fix issues.

FYI: There are a lot of Gazebo plugins in the model that ZeroSim doesn't support out of the box -- so if you need them you may have to implement them yourself.  Would be tremendous if you did implement them if you could contribute them to ZeroSim.

alex-ssom, 2022-04-06:
No problem, that's what open source is for!

We appreciate that you released ZeroSim to the public. I have experience with ROS and linux, but at the moment I have no experience whatsoever with Unity and C#. I have a colleague that's got the opposite set of experiences. It sure is a possibility for us to contribute back to ZeroSim.

alex-ssom, 2022-04-11:
For anyone whom might be interested. At the moment, the `Import URDF` script doesn't handle the `link_origin_rpy` variable. I've had to manually rewrite my xacro files with every `link_origin_rpy` set to `"0 0 0"` to localise the links properly in Unity.

[image]

micahpearlman, 2022-04-11:
@alex-ssom thanks so much!  We will take a look at supporting that in the near future.

micahpearlman, 2022-04-11:
@alex-ssom searching the googles for a definition of "link_origin_rpy" for URDF and not seeing it?  Any references to definition of this in context of URDF?  Seems to be an Melodic xacro conversion issue?

alex-ssom, 2022-04-11:
I agree with you, the documentation of this variable is very sparse. It's mentionned in the [link documentation](https://wiki.ros.org/urdf/XML/link) and in the [Building a Visual Robot Model with URDF from Scratch](http://wiki.ros.org/urdf/Tutorials/Building%20a%20Visual%20Robot%20Model%20with%20URDF%20from%20Scratch#CA-7a98bf230e397bebd5c9a0ac75388002cbaa6457_28) tutorial. In my application, I'm using the `link_origin_rpy` variable to modify the origin of the CAD file.

### xacro output
When using `link_origin_rpy` my urdf is displayed properly in RViz on both melodic and noetic.
[image]
When running 
```bash
rosrun xacro xacro src/leo_common/leo_description/urdf/leo_sim.urdf.xacro > leo_ssom_ROSVERSION_xacro.urdf
```
I get the following output.

**Noetic**
```xml
<?xml version="1.0" ?>
<!-- =================================================================================== -->
<!-- |    This document was autogenerated by xacro from src/leo_common/leo_description/urdf/leo_sim.urdf.xacro | -->
<!-- |    EDITING THIS FILE BY HAND IS NOT RECOMMENDED                                 | -->
<!-- =================================================================================== -->
<robot name="leo_sim">
  <!-- LEO GAZEBO -->
  <link name="base_footprint"/>
  <link name="base_link">
    <inertial>
      <mass value="1.584994"/>
      <origin xyz="-0.019662 0.011643 -0.031802"/>
      <inertia ixx="0.01042" ixy="0.001177" ixz="-0.0008871" iyy="0.01045" iyz="0.0002226" izz="0.01817"/>
    </inertial>
    <visual>
      <geometry>
        <mesh filename="package://leo_description/models/Chassis.dae"/>
      </geometry>
    </visual>
    <collision>
      <geometry>
        <mesh filename="package://leo_description/models/Chassis_outline.stl"/>
      </geometry>
    </collision>
  </link>
  <link name="rocker_L_link">
    <inertial>
      <mass value="1.387336"/>
      <origin xyz="0 0.01346 -0.06506"/>
      <inertia ixx="0.002956" ixy="-0.000001489324" ixz="-0.000008103407" iyy="0.02924" iyz="0.00007112" izz="0.02832"/>
    </inertial>
    <visual>
      <geometry>
        <mesh filename="package://leo_description/models/Rocker.dae"/>
      </geometry>
    </visual>
    <collision>
      <geometry>
        <mesh filename="package://leo_description/models/Rocker_outline.stl"/>
      </geometry>
    </collision>
  </link>
  <link name="rocker_R_link">
    <inertial>
      <mass value="1.387336"/>
      <origin xyz="0 0.01346 -0.06506"/>
      <inertia ixx="0.002956" ixy="-0.000001489324" ixz="-0.000008103407" iyy="0.02924" iyz="0.00007112" izz="0.02832"/>
    </inertial>
    <visual>
      <geometry>
        <mesh filename="package://leo_description/models/Rocker.dae"/>
      </geometry>
    </visual>
    <collision>
      <geometry>
        <mesh filename="package://leo_description/models/Rocker_outline.stl"/>
      </geometry>
[... 2422 lines cut ...]
    </sensor>
  </gazebo>
  <gazebo>
    <light name="left_headlight" type="spot">
      <pose>0.659915 -0.501804 1 0 -0 0</pose>
      <diffuse>0.5 0.5 0.5 1</diffuse>
      <specular>0.1 0.1 0.1 1</specular>
      <direction>0 0 -1</direction>
      <attenuation>
        <range>20</range>
        <constant>0.5</constant>
        <linear>0.01</linear>
        <quadratic>0.001</quadratic>
      </attenuation>
      <cast_shadows>0</cast_shadows>
      <spot>
        <inner_angle>0.6</inner_angle>
        <outer_angle>1</outer_angle>
        <falloff>1</falloff>
      </spot>
    </light>
  </gazebo>
</robot>
```
Let me know if more info is required.

alex-ssom, 2022-04-11:
Sorry for the data dump and the confusion, I'm so used to my tools that I forgot to mention that `link_origin_rpy` is a xacro macro to define the link origin rpy as per the urdf link documentation at:  https://wiki.ros.org/urdf/XML/link
Here's how it's used inside my macro.
```xml
<origin xyz="${link_origin_xyz}" rpy="${link_origin_rpy}"/>
```

I'll try to be as clear as possible, using the `rear_mounting_plate` as an example.
The CAD origin of the `rear_mounting_plate` is the same as the origin of the `front_mounting_plate` but in order to simplify the positioning of accessories on the mounting plate, I wanted to relocate its origin on the center hole at the front of the `rear_mounting_plate`. Then rotate it's origin so that it's properly oriented.
[image] 
For some reason, while this works when displaying the urdf in RViz, it doesn't work when imported in Unity.
```xml
  <link name="rear_mounting_plate_base_link">
    <visual>
      <geometry>
        <mesh filename="package://leo_ssom_addons_description/meshes/visual/RearMountingPlate.dae"/>
      </geometry>
      <origin rpy="0 0 -1.57079632679" xyz="0.155 0 -0.0085"/>
    </visual>
    <collision>
      <geometry>
        <mesh filename="package://leo_ssom_addons_description/meshes/collision/RearMountingPlate.stl"/>
      </geometry>
      <origin rpy="0 0 -1.57079632679" xyz="0.155 0 -0.0085"/>
    </collision>
  </link>
  <joint name="rear_mounting_plate_joint" type="fixed">
    <parent link="base_link"/>
    <child link="rear_mounting_plate_base_link"/>
    <axis xyz="0 0 0"/>
    <origin rpy="0 0 0" xyz="-0.045 0 0.0085"/>
  </joint>
```

When modifying the xacro definition to remove the use of `<origin rpy="X Y Z"/>` I get a properly imported urdf, but the link origins aren't where i'd like them to be.
[image]

```xml
 <link name="rear_mounting_plate_base_link">
    <visual>
      <geometry>
        <mesh filename="./leo_ssom_addons_description/meshes/visual/RearMountingPlate.obj"/>
      </geometry>
      <origin rpy="0 0 0" xyz="0 0 -0.0085"/>
    </visual>
    <collision>
      <geometry>
        <mesh filename="./leo_ssom_addons_description/meshes/collision/RearMountingPlate.obj"/>
      </geometry>
      <origin rpy="0 0 0" xyz="0 0 -0.0085"/>
    </collision>
  </link>
  <joint name="rear_mounting_plate_joint" type="fixed">
    <parent link="front_mounting_plate_base_link"/>
    <child link="rear_mounting_plate_base_link"/>
    <axis xyz="0 0 0"/>
    <origin rpy="0 0 0" xyz="0 0 0"/>
  </joint>
```

I'll formulate it this way. As a user, I'd like to only keep a single urdf definition of my robot that can use the `<origin rpy="X Y Z"/>` argument and that can be imported in Unity without needing modifications.

micahpearlman, 2022-04-11:
Again, apologies, totally lost -- maybe because it's Monday -- super appreciate your patience.  I don't understand what the "good" URDF is and the "bad" URDF is above and what the difference are exactly?  Is it like a 90 degree rotation about an axis?  Or something more? Do you think it would be possible to replicate this with a super simple XAcro file with simple geo? (FYI: I know nearly nothing about XAcro other then how to convert to URDF)

BTW:  If it is an issue with everything is rotated 90 degrees about an axis I suspect it may be an issue in the `convert_meshes_to_obj.sh` (https://github.com/fsstudio-team/zerosim_docker/blob/master/tools/convert_meshes_to_obj.sh) + the `dae_to_obj.py` script (https://github.com/fsstudio-team/zerosim_docker/blob/master/tools/dae_to_obj.py).  It makes some assumptions about the models `up` and `forward` axis, specifically the line: `bpy.ops.export_scene.obj(filepath=fobj, axis_forward='Y', axis_up='Z')` that may be incorrect for your 3D models perhaps?

alex-ssom, 2022-04-12:
Thanks for you patience as well, I'm quite fluent, but English isn't my first language so there might be some stuff that's lost in translation.

FYI: xacro is just an xml macro language on top of the urdf xml. It makes life easier when you want to add links and joints in a modular way.
As an example, this is the file that i use to add parts to the LeoRover original URDF:
```xml
<?xml version="1.0" encoding="utf-8"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">
    <xacro:include filename="$(find leo_ssom_addons_description)/urdf/e_stop.xacro" />
    <xacro:include filename="$(find leo_ssom_addons_description)/urdf/mounting_plates.xacro" />
    <xacro:include filename="$(find leo_ssom_addons_description)/urdf/realsense.xacro" />
    <xacro:include filename="$(find leo_ssom_addons_description)/urdf/jetson_nano.xacro" />
    <xacro:include filename="$(find leo_ssom_addons_description)/urdf/rplidar.xacro" />
</robot>
```
### Link origin issue
I believe you found the root cause of the issue. The link origin rotation and translations values aren't used when importing the URDF. I think that the conversion script is where this should be applied. Probably with `bpy.ops.transform.rotate(rpy)` and `bpy.ops.transform.translate(xyz)`.

I investigated by opening my `.dae` file in Blender.

[image]

I then manually applied the `<origin rpy="0 0 -1.57079632679" xyz="0.155 0 -0.0085"/>` rotations then translations.

[image]

The origin is now where the URDF is expecting it. 
I wonder how hard it would be to add the origin rotation and translation to the conversion script. My guess is that the hardest part would be to load the urdf file and parse it properly.

### Validation with joint origin applied
**After import**
[image]

**After joint origin is applied** the part is properly located and the joint is where it's expected to be.
`<origin rpy="0 0 0" xyz="-0.045 0 0.0085"/>`
[image]