fsstudio-team/ZeroSimROSUnity #26: Moveit not working with UR10

ptiwari0664, 2022-05-06:
Hi,

I am trying to run UR10 with moveit as per the instructions. However, Moveit is not able to connect with Unity and it says 

```
[ WARN] [1651847479.094899661]: Waiting for /follow_joint_trajectory to come up
[ERROR] [1651847491.095258476]: Action client not connected: /follow_joint_trajectory
```

I have tried multiple options with sim param, as its should remap */follow_joint_trajectory* to */arm_controller/follow_joint_trajectory*, but it is not able to do it. Has anybody faced the same issue ?

tejasps28, 2022-05-07:
I'm also facing the same issue!

micahpearlman, 2022-05-09:
@ptiwari0664 & @tejasps28 

It looks like the newer versions of Unity mess up the rigged Unity UR10 model.  You can try to re-import the UR10 URDF or wait for us to update.

tejasps28, 2022-05-09:
> @ptiwari0664 & @tejasps28 
> 
> It looks like the newer versions of Unity mess up the rigged Unity UR10 model.  You can try to re-import the UR10 URDF or wait for us to update. 

Hello, thank you for your reply. To re-import the URDF we'll need to follow your "Import URDF" tutorial right?

micahpearlman, 2022-05-09:
Folllow the "LEO Robot" Example, except for UR10: https://github.com/fsstudio-team/ZeroSimROSUnity#import-urdf

Also, you may need to use a 2020 version of Unity as we have not done any thorough testing with 2021 and it seems to break a lot of things -- we are slowly working through the issues with 2021.

tejasps28, 2022-05-09:
> Folllow the "LEO Robot" Example, except for UR10: https://github.com/fsstudio-team/ZeroSimROSUnity#import-urdf
>
> Also, you may need to use a 2020 version of Unity as we have not done any thorough [...]

Okay, will try this out. Thank you!
Also, yes. I am using Unity version 2020.3.33f1 LTS.

micahpearlman, 2022-05-09:
FYI:  Some random notes for UR10 Import from awhile ago -- should be still relevant.  Note: the ZeroSim Docker image already has UR10 repo in the Catkin Worksapace.

1. Convert STL & DAE meshes to OBJ: `/zerosim_tools/convert_meshes_to_obj.sh ./src/universal_robot/ur_description/meshes/ur10/`
2. Generate UR10 URDF `rosrun xacro xacro src/universal_robot/ur_description/urdf/ur10_robot.urdf.xacro transmission_hw_interface:=hardware_interface/PositionJointInterface > /tmp/ur10.urdf`

3. Copy the generated URDF `docker cp my_zerosim_vnc_docker:/tmp/ur10.urdf .`
4. Copy the meshes: `docker cp my_zerosim_vnc_docker:/catkin_ws/src/universal_robot/ur_description/meshes .`
5. Fixup the mesh paths

tejasps28, 2022-05-09:
@micahpearlman Thanks a lot for this precise guidance. I shall let you know how it goes. Cheers!

tejasps28, 2022-05-09:
> @tejasps28 after further analysis I may be sending you down the wrong rabbit hole.  Apologies.  I am now seeing some weird connection issue through ROS bridge.  In the Unity console I'm getting:
> 
> ```
> ERROR: ZOROSBridgeConnection::ClientReadAsync System.InvalidOperationException: Queue empty.
> ```
> 
> And in the Docker I'm getting:
> 
> ```
> [ERROR] [1652115654.246938824]: Unable to identify any set of controllers that can actuate the specified joints: [ elbow_joint shoulder_lift_joint shoulder_pan_joint wrist_1_joint wrist_2_joint wrist_3_joint ]
> [ERROR] [1652115654.247015161]: Known controllers and their joints:
> ```

Uh, yes. I had noticed these myself too. So could you suggest what's the right way forward?

tejasps28, 2022-05-09:
> Uh, yes. I had noticed these myself too. So could you suggest what's the right way forward?

Also, i did try your mentioned steps and got the same error.

tejasps28, 2022-05-09:
> Not super clear on that at the moment.  Trying to back track to what has changed to have caused this issue -- we haven't done a lot of changes in the past several months, either to the Unity code base or the Docker image -- so it is one of those very odd bugs.

Okay. Please let me know the solution to this when you find the bug. Thank you!

micahpearlman, 2022-05-09:
[image]

So RQT works just fine (see above).  Something broken with the RViz config.  Try launching RViz without the configuration and hand configure it?

tejasps28, 2022-05-11:
@micahpearlman Thanks, i did this and was able to reproduce the same using RQT. Willl try rviz configuration manually and let you know how it goes.

ptiwari0664, 2022-05-22:
@micahpearlman @tejasps28  Hi, I have found the problem. Just correct the ZO_CONTROLLER_MANAGER_SERVICE with name `undefined_10 -> ur_10_control`. It will work. Please correct and commit. 

I have verified it with docker and VNC. I will try to test it on noetic as well.

tejasps28, 2022-05-22:
> @micahpearlman @tejasps28  Hi, I have found the problem. Just correct the ZO_CONTROLLER_MANAGER_SERVICE in Unity Editor with name `undefined_10 -> ur_10_control`. It will work. Please correct and commit. 
> 
> I have verified it with docker and VNC. I will try to test it on noetic as well. 

Yes, i tried this! It seems to work perfectly fine! Thank you @ptiwari0664  @micahpearlman

micahpearlman, 2022-05-23:
Fantastic!  I will put in a fix for this ASAP.

anqitongxue, 2022-10-11:
这里是氨气同学的QQ邮箱，您的邮件我已收到，祝生活愉快~