fsstudio-team/ZeroSimROSUnity #27: Custom Message Creation

tejasps28, 2022-06-07:
I want to work with custom messages and dont know how is it built in zerosim. Is there an option to work with custom messages?
Please help!
Thanks in Advance! @micahpearlman

anqitongxue, 2022-06-07:
这里是氨气同学的QQ邮箱，您的邮件我已收到，祝生活愉快~

tejasps28, 2022-06-07:
> 这里是氨气同学的QQ邮箱，您的邮件我已收到，祝生活愉快~

Im sorry i dont understand.

micahpearlman, 2022-06-07:
@tejasps28 You will want to derive your message from [`ZOROSMessageInterface`](https://github.com/fsstudio-team/ZeroSimROSUnity/blob/68dbfe41dedf8244f244e38c63aedb458bed50d6/Runtime/Scripts/ROS/MessageTypes/ZOROSMessageInterface.cs).  For example:

```
public class ImuMessage : ZOROSMessageInterface {
        [Newtonsoft.Json.JsonIgnore]
        public string MessageType { get { return ImuMessage.Type; } }

        [Newtonsoft.Json.JsonIgnore]
        public static string Type { get { return "sensor_msgs/Imu"; } }

        public HeaderMessage header { get; set; }

        public QuaternionMessage orientation { get; set; }
        public double[] orientation_covariance { get; set; }
        public Vector3Message angular_velocity { get; set; }
        public double[] angular_velocity_covariance { get; set; }
        //  Row major about x, y, z axes
        public Vector3Message linear_acceleration { get; set; }
        public double[] linear_acceleration_covariance { get; set; }
        //  Row major x, y z 

        public ImuMessage() {
            this.header = new HeaderMessage();
            this.orientation = new QuaternionMessage();
            this.orientation_covariance = new double[9];
            this.angular_velocity = new Vector3Message();
            this.angular_velocity_covariance = new double[9];
            this.linear_acceleration = new Vector3Message();
            this.linear_acceleration_covariance = new double[9];
        }

        public ImuMessage(HeaderMessage header, QuaternionMessage orientation, double[] orientation_covariance, Vector3Message angular_velocity, double[] angular_velocity_covariance, Vector3Message linear_acceleration, double[] linear_acceleration_covariance) {
            this.header = header;
            this.orientation = orientation;
            this.orientation_covariance = orientation_covariance;
            this.angular_velocity = angular_velocity;
            this.angular_velocity_covariance = angular_velocity_covariance;
            this.linear_acceleration = linear_acceleration;
            this.linear_acceleration_covariance = linear_acceleration_covariance;
        }

        public void Update() {
            this.header.Update();
        }

    }
```

tejasps28, 2022-06-10:
@micahpearlman yes, i tried it and it works fine! Thanks for the help!

closed 2022-06-10 by micahpearlman