fsstudio-team/ZeroSimROSUnity #28: Unable to import package

vbelkina, 2022-06-10:
Hello, I am trying to follow the tutorial to set this up, however, I keep getting the "Error adding package: git@github.com:fsstudio-team/ZeroSimROSUnity.git"
[image]

I am running on Ubuntu 20.04. I tried a couple of versions of Unity 2020.3.18f1 and 2020.3.35f1 to see if that helped. Not sure if this is something that has happened to others and if there's any solution. Thanks.

micahpearlman, 2022-06-10:
Tryusing https instead of ssh: https://github.com/fsstudio-team/ZeroSimROSUnity.git

If https works, it is probably because you do not have ssh authentication setup for GitHub.

vbelkina, 2022-06-13:
Hi, thanks for your reply. 
The https also did not work:
[image]

I have tried to import other packages, and I don't have any issues.

micahpearlman, 2022-06-13:
@vbelkina  My apologies, it seems to have been an issue on our end.  Please try again.

vbelkina, 2022-06-13:
It worked! Thank you for your help.

closed 2022-06-13 by micahpearlman

vbelkina, 2022-10-11:
Hi, thanks for your reply!
The https also didn't work.