fsstudio-team/ZeroSimROSUnity #29: Unity 2020. add packgae from git url failed

CheAbdullah, 2022-07-16:
[Package Manager Window] Cannot perform upm operation: Unable to add package [git@github.com:fsstudio-team/ZeroSimROSUnity.git]:
  Error when executing git command. Host key verification failed.
  fatal: Could not read from remote repository.

Select Add Package From Git URL... and enter git@github.com:fsstudio-team/ZeroSimROSUnity.git.
{https://github.com/fsstudio-team/ZeroSimROSUnity}

anqitongxue, 2022-07-16:
这里是氨气同学的QQ邮箱，您的邮件我已收到，祝生活愉快~

CheAbdullah, 2022-07-17:
[Package Manager Window] Cannot perform upm operation: Unable to add package [git@github.com:fsstudio-team/ZeroSimROSUnity.git]:
  Error when executing git command. git@github.com: Permission denied (publickey).
  fatal: Could not read from remote repository.

  Please make sure you have the correct access rights
  and the repository exists.

is this link working? git@github.com:fsstudio-team/ZeroSimROSUnity.git

CheAbdullah, 2022-07-17:
DOWNLOAD THIS .JSON.
INSTALL FROM FILE##

https://raw.githubusercontent.com/fsstudio-team/ZeroSimROSUnity/master/package.json

closed 2022-07-17 by CheAbdullah

Rive4, 2023-01-11:
Hello,

I tried in both ways but I am still having problems to import it.
Using package.json file it appears in package manager that I installed the package, but it does not import anything.