fsstudio-team/ZeroSimROSUnity #31: 3D lidar orientation issue

manikjslb, 2022-11-29:
I have been trying to import a 2D and 3D lidar to turtlebot3_waffle using the provided sample turtlebot scene  tutorial.

**What I am trying to do ** is that, if I translate and rotate the turtlebot, then the 3D pointcloud as well as 2D scan data should also translate and rotate similar amount to that of robot base_foorprint.

I am successful in doing so for 2D lidar however, when I import the 3D lidar in scene; then depending upon selected `parent transform id` the 3D pointcloud **ONLY** either rotates or translates.

I have attached some screenshots of Unity editor to show you 3D Lidar's different configuration, screenshot of tf tree and screenshots of foxglove where I am visualizing the data.

`P.S. Observe how 2D scan (white) and 3D pointcloud (green) is aligned or misaligned.` 
**If what I am trying to achieve is successful then the 2D scan and 3D pointcloud would never be misaligned no matter however I move the robot.**

## TF Tree
[image]

## test 1 - `parent transform id = base_footprint`     

| Unity Settings      | Initial Pose | after applying only rotation to base   |
| :---        |    :----:   |          ---: |
| [image]      | [image]        | [image]    |

in this test if I apply only translation then the 2D and 3D pointclouds remains aligned.

## test 2 - `parent transform id = odom`     

| Unity Settings      | Initial Pose | after applying only translation to base  |
| :---        |    :----:   |          ---: |
| [image]      | [image]        | [image]    |

in this test if I apply only rotattion then the 2D and 3D pointclouds remains aligned.

Can someone help me find any bug in my understanding or help me find a solution?

micahpearlman, 2022-11-30:
Thank you for reporting the issue, apologies we will not be able to look into it immediately until probably January.

manikjslb, 2022-11-30:
@micahpearlman thanks you for response.
could you guess where the problem might be coming from? I can try to solve the bug, can you suggest me a start point?