fsstudio-team/ZeroSimROSUnity #32: Fix:  Joint Angles Bug

caiobarrosv, 2023-02-08:
ZeroSim hinge joints limits the joint angle in the range [-180, 180] degrees.

This limitation can be avoided by using Configurable Joints instead of Hinge Joints.
However, instead of implementing the Configurable Joints, check the possibility to use Articulation Body instead.

​
https://docs.unity.cn/Manual/physics-articulations.html

caiobarrosv, 2023-02-09:
Just found out that unity updated this in 2022.2.0 version:

- Physics: Enabled the option to extend angular limits for Hinge Joint.

micahpearlman, 2023-02-09:
@caiobarrosv can you see if ZeroSim works for 2022.2.0 or greater.  One thing is that we want to avoid is supporting non-LTS versions of Unity. As I can see the latest LTS version is 2021.3.8f1.  If ZeroSim works with 2022.2.0 we can add in the README.md that if this issue comes up then update to 2022.2.0 -- but we do not generally support non LTS Unity versions.

caiobarrosv, 2023-02-09:
ZeroSim is working with Unity 2022.2.5f1.

In this version we have Extended Limits option but it is not working properly.
Even if we set the limits to [-360, 360], the spring target position can only be set in the range [-180, 180].

I posted this issue on the Unity forum - [Link](https://forum.unity.com/threads/extend-hinge-joint-limits-to-360-360-not-working.1397992/)

I tried using Motor instead of Spring, but if we set the limits above 180 degrees, the joint will not respect the limitation and will not stop.

caiobarrosv, 2023-02-10:
The articulation body allows joint angles more than 180 and less than -180.
My recomendation is to remove all the hinge joints from robot manipulators in example scenes like the UR10_test.scene and alert users about this joint limitation issue.

[image]

Although the last LST version is the 2021.3.8f1, after this version we had a lot of important updates for Articulated bodies like:

- 2021.1.9: 
  - Physics: Fixed a crash when attempting to access the Rigidbody from the result of a OnCollision callback, when the object we've collided with is a ArticulationBody. The Collision.rigidbody member is now null in this case. (1320314)

- 2021.1.15:
  - Physics: Added: Added a property for retrieving ArticulationBody components during a collision event. Articulation bodies can be retrieved by Collision.articulationBody​
  - Physics: Added: Added a property for retrieving either ArticulationBody or Rigidbody components to collision events under Collision.body.​
  - Physics: Corrected contact event behavior for the ArticulationBody component so that it matches Rigidbody component behavior. (1339908)

- 2021.1.17
  - Physics: Overhauled the articulation gizmos to have better functionality and to better match other Unity gizmos' style
  - Physics: Fixed an Clamp ArticulationBody mass issue when loading data from disk was not within the valid range for the physics engine. (1316719)

- 2021.1.19
  - Physics: Fixed an issue where the Articulation Bodies were not being visualised in the Physics Debugger. (1343929)​

- 2021.1.28
  - Physics: Fixed an issue where the Articulation Drive did not affecting the joint in Articulation Body when the collider volume was very small. (1330968)

- 2021.2.0
  - Physics: Added icon for Articulation Body Anchor Transform tool.
  - Physics: Added units of measurement to the Articulation Body properties in the scripting documentation.
  - Physics: Improved Articulation Body anchor limit gizmos.
  - Physics: Rearranged the ArticulationBody properties. Moved Damping and Friction after Mass.​
  - Physics: Added: Expose ArticulationBody.collisionDetectionMode property to set various collision detection modes, like Discrete, Continuous and etc.​
  - Physics: Added: ForceMode argument to ArticulationBody.Add force and related functions.​
  - Physics: Added: Property for retrieving ArticulationBody components during a collision event. Articulation bodies can be retrieved by Collision.articulationBody.​
  - Physics: Added: Property for retrieving either ArticulationBody or Rigidbody components to collision events under Collision.body.​

- 2021.2.1
  - Physics: Fixed the ArticulationDrive.forceLimit to actually use forces instead of impulses. Projects with custom set force limit should review values, and consider dividing them by the simulation delta. (1370338)

- 2021.2.8
  - Physics: Fixed an issue where multi-selecting Articulation Bodies would set some properties to the same value. (1379145)

- 2021.2.9
  - Physics: Fixed a crash when switching an ArticulationBody to immovable during a collision event. (1338600)

- 2021.2.12
  - Physics: Fixed ArticulationBody.SetJointPositions and similar setter using List.Capacity instead of List.Count to determine the element count and throwing errors because of that. (1347970)

- 2021.3.9
  - Physics: Fixed Articulation Bodies teleporting to NaN coordinates when changing joint type at runtime. (UUM-7418)​

- 2021.3.12
  - Physics: Fixed ArticulationBody indices not matching when using methods like GetJointPositions and GetDriveTargets. ([UUM-1805](https://issuetracker.unity3d.com/issues/backport-data-indices-dont-match-when-using-articulationbody-getdrivetargets-and-getjointpositions))​

- 2021.3.14
  - Physics: Fixed ArticulationBody SetDriveTargets and SetDriveTargetVelocities methods checking List capacity instead of count and improved the error message. ([UUM-12713](https://issuetracker.unity3d.com/issues/supplied-list-capacity-is-being-checked-instead-of-count-when-calling-articulationbody-dot-setdrivertargets-slash-setdrivetargetvelocities))

- 2022.1.0
  - Physics: Improved the Physics Visualiser layout and moved the Rigidbody and Articulation Body Info panels to the Physics Visualiser.​
  - Physics: Updated ArticulationBody.AddForce and other related method Scripting API documentation with information about current PhysX limitations.​
  - Physics: Clamped the Articulation body drive lower and upper limits to each other. (1310004)
 
- 2022.1.23
  - Physics: Fixed ArticulationBody SetDriveTargets and SetDriveTargetVelocities methods checking List capacity instead of count and improved the error message. (UUM-12713)

- 2022.2.0
  - Physics: Added an option to override Layer collisions for Colliders, Rigidbodies, ArticulationBodies and Character Controllers, which allows to fine tune which bodies should collide with what layers.
  - Physics: Added: ArticulationBody.GetJointExternalForces method which allows to get the forces in reduced space needed to counteract forces applied in the general space for your Articulation chain.​
  - Physics: Added: Drive type enumerator for Articulation Body drives.
  - Physics: Added: Expose additional ArticulationDrive setters that allow direct access to the properties.
  - Physics: Added: Expose setting inertia tensor and center of mass for Rigidbody and ArticulationBody components from the Inspector.​
[... cut ...]
  - Fixed Articulation Bodies with spherical joints spinning uncontrollably in some scenarios. (1418715).
  - Physics: Fixed ArticulationBody Anchor rotation values disappearing when Inspector window width is reduced. (1393075)​
  - Physics: Fixed different Force Modes overwriting previously applied forces for Articulation Bodies. (1349587)
 
I would try to keep it as updated as possible if the goal is to use robotic arms with articulated body.
Maybe for mobile robots we can still use regular joints.