fsstudio-team/ZeroSimROSUnity #33: Fix: Communication with ZeroSim is lost when controller_manager node is terminated

caiobarrosv, 2023-02-08:
When the controller manager is launched using ur10_moveit.launch and then terminated, ROS Suite loses communication with Unity.
To re-establish communication, ZeroSim and ROS Suite must be restarted.

micahpearlman, 2023-02-08:
Hmmm, that's odd -- believe this is a new bug.  What docker/ROS version is being used?  Can this be reproduced with the "standard" ZeroSim Melodic docker image?

caiobarrosv, 2023-02-08:
It happens with the "standard" ZeroSim Melodic docker image as well.

If you run ZeroSim with this command:
```bash
docker run -it --rm \
--publish=9090:9090 \
--publish=11311:11311 \
--publish=8083:8083 \
--publish=80:80 \
--publish=5678:5678 \
--name my_zerosim_vnc_docker \
zerodog/zerosim_ros_vnc:latest \
roslaunch zero_sim_ros basic_unity_editor.launch'
```

I can launch MoveIt1 with the ur10_moveit.launch file (you need to comment the line that includes the basic_unity_editor.launch and change the namespace of the controller_manager package for the name defined in the ZO Controller Manager Service).

However, if you try to terminate the node and try to run again, it won't let you connect to ZeroSim again.

This video shows the bug in detail: 
https://youtu.be/5FiDjH1A29M

micahpearlman, 2023-02-09:
Interesting.  Not sure what it could be as when last in active use about 18 months ago it worked.  Hmmm...

micahpearlman, 2023-02-09:
I wonder if it is something particular about the UR10 launch file or configuration in Unity?  Actually we were actively using the xArm not the UR10...